graspit-simulator/graspit #11: Staubli Control

jvarley, 2016-01-13:
Staubli Control Code does not exist in graspit, can we remove:

jvarley@captiva:~/graspit_org/graspit$ grep -ir Staubli .
./ui/ui_mainWindow.h:    QAction *staubliControl;
./ui/ui_mainWindow.h:        staubliControl = new QAction(MainWindowUI);
./ui/ui_mainWindow.h:        staubliControl->setObjectName(QString::fromUtf8("staubliControl"));
./ui/ui_mainWindow.h:        staubliControl->setName("staubliControl");
./ui/ui_mainWindow.h:        Misc->addAction(staubliControl);
./ui/ui_mainWindow.h:        staubliControl->setIconText(QApplication::translate("MainWindowUI", "Staubli Control Panel...", 0, QApplication::UnicodeUTF8));
./ui/ui_mainWindow.h:        staubliControl->setText(QApplication::translate("MainWindowUI", "Staubli Control Panel...", 0, QApplication::UnicodeUTF8));
./ui/mainWindow.cpp:#ifdef STAUBLI_CONTROL_ENABLED
./ui/mainWindow.cpp:#include "staubli/staubliControlDlg.h"
./ui/mainWindow.cpp:  QObject::connect(mUI->staubliControl, SIGNAL(triggered()), 
./ui/mainWindow.cpp:                   this, SLOT(misStaubliControlDlg()));
./ui/mainWindow.cpp:void MainWindow::misStaubliControlDlg()
./ui/mainWindow.cpp:#ifdef STAUBLI_CONTROL_ENABLED
./ui/mainWindow.cpp:  StaubliControlDlg *dlg = new StaubliControlDlg(mWindow);
./ui/mainWindow.ui:        <addaction name="staubliControl" />
./ui/mainWindow.ui:    <action name="staubliControl" >
./ui/mainWindow.ui:        <cstring>staubliControl</cstring>
./ui/mainWindow.ui:        <string>Staubli Control Panel...</string>
./ui/mainWindow.ui:        <string>Staubli Control Panel...</string>
./ui/mainWindow.h:	void misStaubliControlDlg();
Binary file ./.git/index matches
Binary file ./.obj/mainWindow.o matches
Binary file ./.obj/moc_mainWindow.o matches
./graspit-core.pro:#------------------------------------- Staubli Control ----------------
./graspit-core.pro:staubli{
./graspit-core.pro:	DEFINES += STAUBLI_CONTROL_ENABLED
./graspit-core.pro:	SOURCES += Staubli/staubliControlDlg.cpp \
./graspit-core.pro:	HEADERS += Staubli/staubliControlDlg.h \
./graspit-core.pro:	FORMS += Staubli/staubliControlDlg.ui

jvarley, 2016-02-06:
@mateiciocarlie 
I found an older repo which has the arizona and staubli code. 
https://github.com/CURG/graspit_handop
Should I bring these source files into this repo, or should we remove references to them from this repo?

mateiciocarlie, 2016-02-08:
While it's great to have a nice library of robot controllers in graspIt!, the way it's done here is really not right - it probably should be all plugin based, and core graspit files should not have all these kinds of references. So I agree - until we have time to check this, and re-architect it properly, let's just clean it up...

jvarley, 2016-11-15:
done.

closed 2016-11-15 by jvarley