graspit-simulator/graspit #44: Error saving image

belledon, 2016-06-11:
I apologize is this isn't the proper place to present this (I'm new to github). I have cloned and built Graspit but receive a SIGSEGV segmentation error any time I try to "save image". I also receive the following warning: "Coin warning in glxglue_context_create_software(): Couldn't create GLX context." It could very well be an issue with my setup. If you would be interested in investigating further, I can post the gdb backtrace.

jvarley, 2016-06-13:
Hello,

Thanks for reporting the issue.  I couldn't exaclty replicate your error, but the image saving code was not working for me either. Hopefully this fixes your problem as well.

Please see:
https://github.com/graspit-simulator/graspit/pull/45

Let me know if you still have any problems with saving an image.
-Jake

belledon, 2016-06-13:
Thanks for the quick response! Unfortunately this doesn't prevent the error. However, I am running it on ubuntu 14.04. I will try it again on 12.04 and let you know.

jvarley, 2016-06-13:
can you verify that opengl is installed correctly?

belledon, 2016-06-13:
I can run glxgears and the version info is:

server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.3.0 NVIDIA 340.96
OpenGL core profile shading language version string: 4.30 NVIDIA via Cg compiler
OpenGL version string: 4.4.0 NVIDIA 340.96
OpenGL shading language version string: 4.40 NVIDIA via Cg compiler

I have tried switching over to the integrated driver but it still gave me the same error

jvarley, 2016-06-13:
I just ran it on 2 additional ubuntu 14.04 machines, and it worked  on both.  One had the nvidia 340.96 driver and the other had the nvidia 352.39 driver.  

For the one with the 340.96 driver I ran:
glxinfo | grep -i version
and got:
server glx version string: 1.4
client glx version string: 1.4
GLX version: 1.4
OpenGL core profile version string: 4.3.0 NVIDIA 340.96
OpenGL core profile shading language version string: 4.30 NVIDIA via Cg compiler
OpenGL version string: 4.4.0 NVIDIA 340.96
OpenGL shading language version string: 4.40 NVIDIA via Cg compiler

When I run the one with the 340 driver it does print:
Couldn't create GLX context
But it still saves the image.  

In order to run it, I do the following
cd ~
git clone git@github.com:graspit-simulator/graspit.git
cd graspit
export GRASPIT=$PWD
mkdir build
cd build
cmake ..
make -j5
./graspit_simulator -w plannerMug

At this point graspit should be running with a barrett hand and a mug visible on screen.  Then I do save image.  And there should be a saved image in ~/graspit/images/IMAGE_NAME.jpg

JenniferBuehler, 2016-06-13:
Confirmed I don't get the problem either on 14.04.
The problem with creating the GL context (if it causes the issue at all) should be happening in the Coin SoOffscreenRenderer::renderFromBase() when trying to create the GL context (it's called from render(), which returns a boolean btw, maybe this should be caught as well at some stage @jvarley ?).

If you're happy to edit&recompile the code, and it still happens after newest changes by @jvarley, could you check if it crashes in ivmgr.cpp at myRenderer->render() (in IVmgr::saveImage(), should be line 1458 in src/ivmgr.cpp)?
To keep things simple, just add print statements, eg. ``std::cout<<"OK line "<<__LINE__<<std::endl`` throughout the whole function and see which one gets printed last (saves time not needing to run gdb/valgrind). This should help us find the problem.

Other quick check: Which version of coin are you using? Presuming you installed libcoin80-dev?
What does ``apt-cache show libcoin80-dev`` print?

belledon, 2016-06-14:
Sorry for the late reply. @jvarley, after using make -j5 it worked! And yes I am using libcoin80. Thank you both for your input. I still receive the warning from Coin but I do not see it affecting Graspit

jvarley, 2016-06-14:
great!

closed 2016-06-14 by jvarley