graspit-simulator/graspit #54: Crash on callling GWS projection and Quality Measures

Shashank35, 2016-06-28:
System:
Ubuntu 14.04 LTS , 64 bit

On calling GWS projection and Quality Measures feature on dlr_flask world, Crash observed with the following logs from core dump:

Program received signal SIGSEGV, Segmentation fault.
__sigsetjmp () at ../sysdeps/x86_64/setjmp.S:26
26	../sysdeps/x86_64/setjmp.S: No such file or directory.
(gdb) bt
#0  __sigsetjmp () at ../sysdeps/x86_64/setjmp.S:26
#1  0x00007ffff7982b9f in GWS::buildHyperplanesFromWrenches (this=this@entry=0x1fa6290, 
    wr=wr@entry=0x1fa1050, numWrenches=numWrenches@entry=48, useDimensions=...)
    at /home/nova/Qt_Projects/graspit/src/gws.cpp:470
#2  0x00007ffff7983468 in L1GWS::build (this=0x1fa6290, useDimensions=...)
    at /home/nova/Qt_Projects/graspit/src/gws.cpp:637
#3  0x00007ffff7972736 in Grasp::updateWrenchSpaces (this=this@entry=0x1535e30, useDimensions=...)
    at /home/nova/Qt_Projects/graspit/src/grasp.cpp:356
#4  0x00007ffff7974be8 in Grasp::update (this=0x1535e30, useDimensions=...)
    at /home/nova/Qt_Projects/graspit/src/grasp.cpp:326
#5  0x00007ffff79cfbf7 in World::updateGrasps (this=0x1293080)
    at /home/nova/Qt_Projects/graspit/src/world.cpp:1570
#6  0x00007ffff7a87754 in MainWindow::graspQualityMeasures (this=0x818be0)
    at /home/nova/Qt_Projects/graspit/ui/mainWindow.cpp:673
#7  0x00007ffff753387a in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#8  0x00007ffff68b2a62 in QAction::triggered(bool) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#9  0x00007ffff68b4433 in QAction::activate(QAction::ActionEvent) ()
   from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#10 0x00007ffff6ce2509 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#11 0x00007ffff6ce6a39 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#12 0x00007ffff690851a in QWidget::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#13 0x00007ffff6cea83b in QMenu::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#14 0x00007ffff68b8e2c in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#15 0x00007ffff68bf5dd in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#16 0x00007ffff751f4dd in QCoreApplication::notifyInternal(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#17 0x00007ffff68bed93 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#18 0x00007ffff6933d1c in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#19 0x00007ffff6933289 in QApplication::x11ProcessEvent(_XEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#20 0x00007ffff695ab32 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#21 0x00007ffff3121e04 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#22 0x00007ffff3122048 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#23 0x00007ffff31220ec in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#24 0x00007ffff754c7a1 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
    () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#25 0x00007ffff695abe6 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#26 0x00007ffff751e0af in QEventLoop::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#27 0x00007ffff751e3a5 in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#28 0x00007ffff7523b79 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#29 0x00007ffff7979237 in GraspItGUI::startMainLoop (this=this@entry=0x7fffffffdd50)
    at /home/nova/Qt_Projects/graspit/src/graspitGUI.cpp:281
#30 0x0000000000401958 in main (argc=1, argv=0x7fffffffde88)
    at /home/nova/Qt_Projects/graspit/src/main.cpp:98

Tried to debug the same but no clue how to go about it

Thanks in advance

jvarley, 2016-06-28:
Can I have a bit more information about what you do leading up to the segfault. 

If I run the following:
./graspit-simulator
Open the dlr flask work
optionally call grasp->autograsp
grasp-> create GWS Projection
select 3 fixed values
click ok.

I get a GWS projection and it looks fine.  I tested this on both 12.04 and 14.04.  Can you  verify that the following has been run:
sudo apt-get install libqhull-dev

Shashank35, 2016-06-28:
Thanks for the quick reply. Yes qhull was installed properly and I followed the same steps when I got this crash.
From what I understand it is library incompatibility issue with the 64 bit machine. I created a new 32 bit virtual machine with Ubuntu 12.02 and everything works fine (I can see GWS projection and Quality measure). 
Let me know if I can provide any other detail.

jvarley, 2016-06-28:
Unfortunately, this is a bit difficult for me to debug as I don't have a machine that I can replicate the error with.  

It appears you are dying on ~line 470 from gws.cpp
if ((exitcode = setjmp(qh errexit))) {

Where it makes the call to setjmp.  Basically this is where is calls qhull to create the convex hull for the grasp wrench space.  For whatever reason the setjmp call does not seem to work on your 64 bit machine. 

Try writing a small test program to use setjmp and see if that works for you?
for example:
https://en.wikipedia.org/wiki/Setjmp.h

Shashank35, 2016-06-29:
I tried the test program and it works fine.

On analyzing the threads further the issue looks somewhat similar to this:
https://github.com/ivmai/bdwgc/issues/86
I will dig deeper and post the results soon.

jvarley, 2016-06-30:
Ok, please post if you find anything out.

jvarley, 2016-07-06:
Any luck with this?

closed 2017-07-28 by mateiciocarlie