graspit-simulator/graspit #68: exporting more grasp information

qqwan, 2016-10-03:
Hi,

I'd like to pull out the following items for each grasp, and preferably save them into a txt file (like the way hand position and grasp quality is saved when running a planner), and I was just wondering if you can point me to the right functions to use for each parameter. 

- the joint positions of the fingers
- the contact location and wrench in hand frame for each contact

thanks!

jvarley, 2016-10-04:
I would possibly recommend the dof values instead of the joint values. you can get them with:

If you just want to hack the EGPlanner in GraspIt!, then you can just add the following lines to get the dof values:
double dof[mHand->getNumDOF()];
mHand->getDOFVals(dof);

And extend the graspRecord class to write whatever information you would like:
https://github.com/graspit-simulator/graspit/blob/master/src/graspRecord.cpp

If you run it with the EGPlannerDlg, then just iterate over the list of grasps in:
https://github.com/graspit-simulator/graspit/blob/master/ui/EGPlanner/egPlannerDlg.cpp#L568

and export them to a file.

If you are doing this with ROS, then take a look at:
https://github.com/CURG/graspit_interface/blob/master/src/graspit_interface.cpp#L783
I exposed some common graspIt! functionality via ros services and an action server for the planner.  
There is a python client:
https://github.com/CURG/graspit_commander

jvarley, 2016-10-04:
In order to get contact locations:  
There is alot done with the contacts in the various energy functions for the EGPlanner so it maybe worth poking around in there to see what sorts of things are being done. 

For example:
https://github.com/graspit-simulator/graspit/blob/master/src/EGPlanner/energy/contactEnergy.cpp#L21

Shows how to collect the contacts, and if you want to get the contacts in the frame of the body:
https://github.com/graspit-simulator/graspit/blob/master/include/contact/contact.h#L240

You can get the body pose  and link of the hand in the world frame using:
https://github.com/graspit-simulator/graspit/blob/master/include/contact/contact.h#L231

So you can get the pose of the contact in the world frame using:
 loc * body1->getTran();
Where loc is the location of the contact in the body1 frame of reference.  Then from the hand you can call getBase.getTran() to get the pose of the base of the hand in the world frame.

qqwan, 2016-10-04:
This is very helpful, thanks! 

Also, do you have any example files for manually indicating the sampling locations/patterns for the planner?

jvarley, 2016-10-04:
The EGPlanner uses simulated annealing to sample the planning space.  Take a look at the list planner if you want to create a list of grasps and then have graspit report the quality for them.  I have been playing around a bit with the list planner Matei originally wrote which sounds like what you are looking for here:
https://github.com/jvarley/graspit/tree/list_planner

Run graspit/ 
File-> Open -> plannerMug.xml
Grasp-> ListPlanner 

I never quite finished this PR, but it should give you many of the components you are after.

jvarley, 2016-10-04:
Perhaps you could clean this up a bit and finish the PR?:
https://github.com/graspit-simulator/graspit/pull/69

closed 2017-02-03 by jvarley