graspit-simulator/graspit #70: CMake failed

MrGba2z, 2016-11-03:
Hi there,

I was trying to install graspit! on my virtualbox(Ubuntu16.04.01)

I followed the manual and got following errors.

```
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Qhull: /usr/lib/x86_64-linux-gnu/libqhull.so  
-- QHULL found (include: /usr/include;/usr/include/libqhull, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- Looking for SoQt headers -- found /usr/include/Inventor/Qt/SoQt.h
-- Looking for SoQt library -- found /usr/lib/x86_64-linux-gnu/libSoQt4.so
-- Looking for SoQt library debug -- not found
-- Result of soqt_config: -lSoQt -lQtOpenGL -lQtGui -lQtCore -lXmu -lXi -lCoin -lGL -lXext -lSM -lICE -lX11 -ldl -lpthread

-- soqt_config failed? 0
-- Looking for sgemm_
-- Looking for sgemm_ - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- A library with BLAS API found.
-- A library with BLAS API found.
-- Looking for cheev_
-- Looking for cheev_ - not found
CMake Error at /usr/share/cmake-3.5/Modules/FindLAPACK.cmake:342 (message):
  A required library with LAPACK API not found.  Please specify library
  location.
Call Stack (most recent call first):
  CMakeLists.txt:44 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu/graspit/build/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu/graspit/build/CMakeFiles/CMakeError.log".
```

Am I missing something? Or any help?

Thanks

jvarley, 2016-11-03:
try: 

sudo apt-get install liblapack-dev

MrGba2z, 2016-11-03:
problem solved! Thanks.

closed 2016-11-03 by MrGba2z