graspit-simulator/graspit #71: Unable to import worlds

MrGba2z, 2016-11-04:
Hi there,

I followed manual (get started chapter) to import dlr_flask.xml. But nothing shows up in my window. Am I missing something or is the manual out of date?

Thanks

environment: Ubuntu LTS 16.04.01
[image]

jvarley, 2016-11-04:
In the terminal you are running GraspIt! is anything printed out when you run launch graspit and open the world?

Also in that same terminal, could you please run:
```
env | grep GRASPIT
```

MrGba2z, 2016-11-04:
Oh, I did't try to run it in terminal.  
The environment setting (GRASPIT) is ok. It's set to the graspit home path correctly.

Everything is good when I start it in command line.(already solves my problem)  
But when I double click the application in the GUI. It raises my problem.

jvarley, 2016-11-04:
What GUI are you referring to, GraspIt! or an IDE?

MrGba2z, 2016-11-04:
Oh, I am talking about the ubuntu GUI.
What I did is just double click the executable graspit_simulator in /build folder.
[image]

jvarley, 2016-11-04:
Got it. Yes, it is not working because the GRASPIT environment variable is not set when you launch it this way.  Graspit uses that environment variable to locate the worlds, and models subdirectories.  I would recommend launching it from the terminal, because you will also be able to see any output messages from Graspit while it is running.

jvarley, 2016-11-04:
If you really want to run it from the GUI, then make a small bash script and place it next to graspit_simulator call it run.sh:

#!/bin/bash                                                                     
export GRASPIT=/home/jvarley/graspit
./graspit_simulator

MrGba2z, 2016-11-04:
Thanks for the help!

closed 2016-11-04 by MrGba2z

Elmigdad, 2018-03-04:
hello 
how do i open Graspit! via terminal!?
and how to create my own model

Elmigdad, 2018-04-06:
hello 
I have created my robot hand, but there is an issue of miss matching between geometry and axis of rotation .. how do I fix it please ???