graspit-simulator/graspit #72: dynamics example

MrGba2z, 2016-11-10:
Hi,
I followed manual to run the dynamics example. However I got hundreds of following errors and the machine is close to freezing.

```
Coin warning in SbRotation::setValue(): rotateTo argument has zero length.
Coin warning in SbRotation::setValue(): rotateTo argument has zero length.
Coin warning in SbRotation::setValue(): rotateTo argument has zero length.
Coin warning in SbRotation::setValue(): rotateTo argument has zero length.
```

Btw, I assume (pop state, push state, start simulation) is the 'dynamic tool bar' mentioned in the manual.

jvarley, 2016-11-10:
I am not sure how to replicate your problem.  Can you give more information in regards to what world you are loading, and how you are running it.

MrGba2z, 2016-11-10:
Following is what I did:

1. open graspit simulator from command line
2. open the world 'barrettGlassDyn.xml'
3. click 'start simulation' button

jvarley, 2016-11-14:
And then your whole machine crashes?  I am still on 14.04 atm.  I will try and get a hold of a 16.04 machine and test this.

MrGba2z, 2016-11-14:
It's like a dead loop, making the machine almost freezing. But I still can kill it.

jvarley, 2016-11-15:
can you try this with the plannerMug world.

MrGba2z, 2016-11-15:
plannerMug is fine but it does nothing after click the 'start simulation' button.(no errors in console either)

jvarley, 2016-11-15:
click Grasp->autoGrasp after starting the simulation

MrGba2z, 2016-11-15:
After clicking autoGrasp, it closes the hand and throw three lines of 'Coin warning in SbRotation::setValue(): rotateTo argument has zero length.' Then it stops.

Is that the expected behavior?

jvarley, 2016-11-15:
Is there a difference in the speed at which the hand closes compared to when physics is not enabled.  This sounds like the expected behaviour. The physics engine will automatically stop once the grasp is finished, or the dynamics engine becomes unstable.

MrGba2z, 2016-11-15:
They do have a speed difference. And what about barrettGlassDyn.xml?

BTW: Is there any robot with other tactile sensor other than filtered sensor(which I found in BarrettBH8_280_Tactile) What I' looking for is a force/pressure tactile sensor. If it is off-topic, I can open a new issue.

jvarley, 2016-11-15:
It sounds like the dynamics engine is working as expected.  

As far as force/pressure sensing.  The value for the tactile on the filtered sensor is computed at:
https://github.com/graspit-simulator/graspit/blob/master/src/bodySensor.cpp#L156

Currently it is either 0 or 1, but you could easily change:
forces[2]=1.0 to 
forces[2] = contactForce

Or model the pressure/contact however you like here.  

Note, if you do this, the contacts may not light up because the coloring is done for the range 0-1 and the forces are probably much smaller than that.

closed 2017-02-03 by jvarley