graspit-simulator/graspit #73: Compiling on Win10 by using vs2010

tlsdba, 2016-11-14:
Hi, 
I was trying to compile graspIt! by using vs2010. I have compiled Qt 4.7.4, Coin3D, SoQt 1.5.0 and CLAPACK succeed with vs2010, but I failed when I try to compile graspIt! I got following error:

`1>graspitCore.obj : error LNK2005: "class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > __cdecl cmdline::detail::readable_typename<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > >(void)" (??$readable_typename@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@detail@cmdline@@YA?AV?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@XZ) already defined in graspitParser.obj` 
`1>dynamics.obj : error LNK2019: unresolved external symbol "_dgemm_"，referenced in function "_dgemm"`
`1>gloveInterface.obj : error LNK2001: unresolved external symbol "_dgemm_"`
`1>lmiOptimizer.obj : error LNK2001: unresolved external symbol "_dgemm_"`
`1>dynamics.obj : error LNK2019: unresolved external symbol "_dcopy_"，referenced in function "_dcopy"`
`1>grasp.obj : error LNK2001: unresolved external symbol "_dcopy_"`
`1>lmiOptimizer.obj : error LNK2001: unresolved external symbol "_dcopy_"`
`1>maxdet_src.obj : error LNK2001: unresolved external symbol "_dcopy_"`
`1>dynamics.obj : error LNK2019: unresolved external symbol "_dgemv_"，referenced in function "_dgemv"`
`1>lmiOptimizer.obj : error LNK2001: unresolved external symbol "_dgemv_"`
`1>maxdet_src.obj : error LNK2001: unresolved external symbol "_dgemv_"`
`1>dynamics.obj : error LNK2019: unresolved external symbol "_dscal_"，referenced in function "_dscal"`
`1>lmiOptimizer.obj : error LNK2001: unresolved external symbol "_dscal_"`
`1>maxdet_src.obj : error LNK2001: unresolved external symbol "_dscal_"`
`1>dynamics.obj : error LNK2019: unresolved external symbol "daxpy_"，referenced in function "_daxpy"`
`1>lmiOptimizer.obj : error LNK2001: unresolved external symbol "_daxpy_"`
`1>maxdet_src.obj : error LNK2001: unresolved external symbol "_daxpy_"`
`1>ivmgr.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: class QImage __thiscall QGLWidget::grabFrameBuffer(bool)" (__imp_?grabFrameBuffer@QGLWidget@@QAE?AVQImage@@_N@Z), referenced in function "public: void __thiscall IVmgr::saveImage(class QString)" (?saveImage@IVmgr@@QAEXVQString@@@Z)`
`1>lmiOptimizer.obj : error LNK2019: unresolved external symbol "_ddot_"，referenced in function "_ddot"`
`1>maxdet_src.obj : error LNK2001: unresolved external symbol "_ddot_"`
`1>maxdet_src.obj : error LNK2019: unresolved external symbol "_dnrm2_"，referenced in function "_dnrm2"`
`1>E:\Project\graspIt\graspit-master\build\Debug\graspit.dll : fatal error LNK1120: 8 unresolved externals`

Because 'cmdline.h' has error with '#include <cxxabi.h>', so I use this file to replace it [https://dl.dropboxusercontent.com/u/56184755/cmdline.h](url). I'm not sure it is correct.

 I'm really don't know the link error about "_dgemm_", or "_dcopy_". 

Thanks!

jvarley, 2016-11-14:
Hi,
Thanks for the feedback.  Can you please make a Pull Request with the new cmdline.h file. 
https://github.com/graspit-simulator/graspit/pulls

Also we have documentation for compiling GraspIt! on windows at:
http://graspit-simulator.github.io/build/html/installation_windows.html

I am curious why you followed the old manual rather than the new one.  If there is anything that should be clarified in the new one, if so, please make PR's to that as well:
https://github.com/graspit-simulator/graspit-simulator.github.io_src/blob/master/source/installation_windows.rst

Thanks,
-Jake

tlsdba, 2016-11-15:
Thank you for your reply. 
I have some other codes and dependent libraries that compiled by CMake and vs2010. I also try to compile graspIt-2.2 followed the old manual by using CMake and vs2010, but after executing `qmake -t vcapp -o graspit.vcproj graspit.pro` and creating Visual Studio project file, I failed when I open graspit.vcproj. The error is:
`The following error has occurred during XML parsing: File: E:\Project\graspIt\Graspit\graspit.vcproj Line: 5 Column: 21 Error Message: invalid pointer The file 'E:\Project\graspIt\Graspit\graspit.vcproj' has failed to load.`

Thanks!
-Nick

jvarley, 2016-11-15:
yes, the old instructions used qmake rather than cmake to build the project.  There is no graspit.pro file anymore, it was the equivalent of the CMakeLists.txt.  The old instructions will only work with the version of graspit on sourceforge.  I would highly recommend the version of graspit on github though as there have been a good number of improvements to the code base since it has been moved to github.  

I am glad you got it to work.  If you end up with any suggestions for the documentation on how to install on windows, please let me know.

closed 2017-07-28 by mateiciocarlie