graspit-simulator/graspit #101: Qt Application error

disdaining, 2017-03-13:
Database connected. Cannot get rid of the following exception error:

```
Using default inertia matrix
Using default center of gravity
Database successfully opened.
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_S_construct null not valid
Aborted
```

Any idea?

jvarley, 2017-03-13:
See:
https://github.com/graspit-simulator/graspit/pull/26

disdaining, 2017-03-14:
Added all updated files. still not working. :(

```
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_S_construct null not valid
Aborted (core dumped)
```
Any suggestion?

jvarley, 2017-03-20:
graspitGUI.h was refactored and most of its functionality now lives in graspitCore, but they aren't exactly drop in replacements for each other.  

I believe the most relevant changes from that PR to fix your issue are the ones to the files in
src/DBase/DBPlanner/database.h 
src/DBase/DBPlanner/sql_database_manager.cpp 

Can you run GraspIt! via gdb and paste the backtrace.

closed 2017-07-28 by mateiciocarlie

disdaining, 2017-08-22:
Sorry for late response. Here's the trace from the gdb. 
Using default inertia matrix
Using default center of gravity
Database successfully opened.
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid

Thread 1 "graspit_simulat" received signal SIGABRT, Aborted.
0x00007ffff5d74428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) disass
Dump of assembler code for function __GI_raise:
   0x00007ffff5d743f0 <+0>:	mov    %fs:0x2d4,%ecx
   0x00007ffff5d743f8 <+8>:	mov    %fs:0x2d0,%eax
   0x00007ffff5d74400 <+16>:	movslq %eax,%rsi
   0x00007ffff5d74403 <+19>:	test   %esi,%esi
   0x00007ffff5d74405 <+21>:	jne    0x7ffff5d74438 <__GI_raise+72>
   0x00007ffff5d74407 <+23>:	mov    $0xba,%eax
   0x00007ffff5d7440c <+28>:	syscall 
   0x00007ffff5d7440e <+30>:	mov    %eax,%ecx
   0x00007ffff5d74410 <+32>:	mov    %eax,%fs:0x2d0
   0x00007ffff5d74418 <+40>:	movslq %eax,%rsi
   0x00007ffff5d7441b <+43>:	movslq %edi,%rdx
   0x00007ffff5d7441e <+46>:	mov    $0xea,%eax
   0x00007ffff5d74423 <+51>:	movslq %ecx,%rdi
   0x00007ffff5d74426 <+54>:	syscall 
=> 0x00007ffff5d74428 <+56>:	cmp    $0xfffffffffffff000,%rax
   0x00007ffff5d7442e <+62>:	ja     0x7ffff5d74450 <__GI_raise+96>
   0x00007ffff5d74430 <+64>:	repz retq 
   0x00007ffff5d74432 <+66>:	nopw   0x0(%rax,%rax,1)
   0x00007ffff5d74438 <+72>:	test   %ecx,%ecx
   0x00007ffff5d7443a <+74>:	jg     0x7ffff5d7441b <__GI_raise+43>
   0x00007ffff5d7443c <+76>:	mov    %ecx,%edx
   0x00007ffff5d7443e <+78>:	neg    %edx
   0x00007ffff5d74440 <+80>:	and    $0x7fffffff,%ecx
   0x00007ffff5d74446 <+86>:	cmove  %esi,%edx
   0x00007ffff5d74449 <+89>:	mov    %edx,%ecx
   0x00007ffff5d7444b <+91>:	jmp    0x7ffff5d7441b <__GI_raise+43>
   0x00007ffff5d7444d <+93>:	nopl   (%rax)
   0x00007ffff5d74450 <+96>:	mov    0x38ea21(%rip),%rdx        # 0x7ffff6102e78
   0x00007ffff5d74457 <+103>:	neg    %eax
   0x00007ffff5d74459 <+105>:	mov    %eax,%fs:(%rdx)
   0x00007ffff5d7445c <+108>:	mov    $0xffffffff,%eax
   0x00007ffff5d74461 <+113>:	retq   
End of assembler dump.
I tested on both linux mint 18.2 and Ubuntu 16.04. Previously i don't know how it solved but now again crashes.

jvarley, 2017-08-22:
can you run bt rather than disass

disdaining, 2017-08-22:
It's already there.
(gdb) **### bt**
#0  0x00007ffff5d74428 in __GI_raise (sig=sig@entry=6)
    at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff5d7602a in __GI_abort () at abort.c:89
Backtrace stopped: Cannot access memory at address 0x7fffffffdcc8

jvarley, 2017-08-23:
Can you try running the following in gdb:
```
thread apply all bt
```

I don't really know how to help you.  I don't have a CGDB setup currently, and there isn't much I can go off of from the back trace you are providing.  I have personally moved away from using CGDB as I found it really difficult to debug in situations like you are experiencing now.  

When  I want to store a large number of grasps, I have been interacting with graspit via:
https://github.com/graspit-simulator/graspit_commander

And then from python using a mongoDB client to store my planned grasps.

jvarley, 2017-08-23:
have you installed the following:
sudo apt-get install libqt4-sql-psql

disdaining, 2017-08-25:
After using:
`thread apply all bt `
I get the following trace results:

```
Database successfully opened.
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid

Thread 1 "graspit_simulat" received signal SIGABRT, Aborted.
0x00007ffff5d74428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) thread apply all bt

Thread 4 (Thread 0x7fffd3fff700 (LWP 3671)):
#0  0x00007ffff5e3a70d in poll () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007ffff2c4a38c in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff2c4a712 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007fffeb21f9d6 in ?? () from /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#4  0x00007ffff2c70bb5 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff3ea26ba in start_thread (arg=0x7fffd3fff700) at pthread_create.c:333
#6  0x00007ffff5e463dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 3 (Thread 0x7fffd8c71700 (LWP 3670)):
#0  0x00007ffff5e3a70d in poll () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007ffff2c4a38c in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff2c4a49c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff2c4a4d9 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff2c70bb5 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff3ea26ba in start_thread (arg=0x7fffd8c71700) at pthread_create.c:333
#6  0x00007ffff5e463dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 2 (Thread 0x7fffe15d7700 (LWP 3669)):
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007ffff74103a6 in QWaitCondition::wait(QMutex*, unsigned long) ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#2  0x00007ffff513db0f in SoQtSignalThread::run() () from /usr/lib/x86_64-linux-gnu/libSoQt.so.20
#3  0x00007ffff740fe3c in ?? () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#4  0x00007ffff3ea26ba in start_thread (arg=0x7fffe15d7700) at pthread_create.c:333
#5  0x00007ffff5e463dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 1 (Thread 0x7ffff7fb4780 (LWP 3665)):
#0  0x00007ffff5d74428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff5d7602a in __GI_abort () at abort.c:89
#2  0x00007ffff63ae84d in __gnu_cxx::__verbose_terminate_handler() ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff63ac6b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff63ac701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff63ac969 in __cxa_rethrow () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff751e6ee in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#7  0x00007ffff75244b9 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#8  0x00007ffff797c2fa in GraspitCore::startMainLoop (this=this@entry=0x7fffffffde70)
    at /home/research/graspit/src/graspitCore.cpp:290
---Type <return> to continue, or q <return> to quit---
#9  0x000000000040230d in main (argc=1, argv=0x7fffffffe008) at /home/research/graspit/src/main.cpp:110
```
Yes, its a bit terrible dealing with CGDB and solving this issue. I was looking at one of the solution at stackoverlow regarding similar error, and the solution was to recompile on the original system rather than the client because the files doesn't exist on the client.

disdaining, 2017-08-25:
Yes i have all the required libraries. I don't know how to use graspit_commander. When i try to build using ccmake .. i get the following error:
```
CMake Error at CMakeLists.txt:4 (find_package):
   By not providing "Findcatkin.cmake" in CMAKE_MODULE_PATH this project has
   asked CMake to find a package configuration file provided by "catkin", but
   CMake did not find one.

   Could not find a package configuration file provided by "catkin" with any
   of the following names:

     catkinConfig.cmake
     catkin-config.cmake

   Add the installation prefix of "catkin" to CMAKE_PREFIX_PATH or set
   "catkin_DIR" to a directory containing one of the above files.  If "catkin"
   provides a separate development package or SDK, be sure it has been
   installed.

```

disdaining, 2017-08-25:
Even on graspit v3.0 i got similar error:
```

Database successfully opened.
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'std::logic_error'
  what():  basic_string::_M_construct null not valid

Thread 1 "graspit_simulat" received signal SIGABRT, Aborted.
0x00007ffff5d67428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) thread apply all bt

Thread 9 (Thread 0x7fffd15fa700 (LWP 6371)):
#0  syscall () at ../sysdeps/unix/sysv/linux/x86_64/syscall.S:38
#1  0x00007ffff2c81cfa in g_cond_wait_until () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff2c11999 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff2c11fbb in g_async_queue_timeout_pop () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff2c6460a in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff2c63bb5 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#6  0x00007ffff31256ba in start_thread (arg=0x7fffd15fa700) at pthread_create.c:333
#7  0x00007ffff5e393dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 4 (Thread 0x7fffd3fff700 (LWP 6366)):
#0  0x00007ffff5e2d70d in poll () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007ffff2c3d38c in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff2c3d712 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007fffeb2129d6 in ?? () from /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0
#4  0x00007ffff2c63bb5 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff31256ba in start_thread (arg=0x7fffd3fff700) at pthread_create.c:333
#6  0x00007ffff5e393dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 3 (Thread 0x7fffd8c64700 (LWP 6365)):
#0  0x00007ffff5e2d70d in poll () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007ffff2c3d38c in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#2  0x00007ffff2c3d49c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#3  0x00007ffff2c3d4d9 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#4  0x00007ffff2c63bb5 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#5  0x00007ffff31256ba in start_thread (arg=0x7fffd8c64700) at pthread_create.c:333
#6  0x00007ffff5e393dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 2 (Thread 0x7fffe15ca700 (LWP 6363)):
#0  pthread_cond_wait@@GLIBC_2.3.2 () at ../sysdeps/unix/sysv/linux/x86_64/pthread_cond_wait.S:185
#1  0x00007ffff74033a6 in QWaitCondition::wait(QMutex*, unsigned long) ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#2  0x00007ffff5130b0f in SoQtSignalThread::run() () from /usr/lib/x86_64-linux-gnu/libSoQt.so.20
#3  0x00007ffff7402e3c in ?? () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#4  0x00007ffff31256ba in start_thread (arg=0x7fffe15ca700) at pthread_create.c:333
#5  0x00007ffff5e393dd in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109

Thread 1 (Thread 0x7ffff7fb3780 (LWP 6359)):
---Type <return> to continue, or q <return> to quit---
#0  0x00007ffff5d67428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
#1  0x00007ffff5d6902a in __GI_abort () at abort.c:89
#2  0x00007ffff63a184d in __gnu_cxx::__verbose_terminate_handler() ()
   from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ffff639f6b6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ffff639f701 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ffff639f969 in __cxa_rethrow () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff75116ee in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#7  0x00007ffff75174b9 in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#8  0x00007ffff796beca in GraspitCore::startMainLoop (this=this@entry=0x7fffffffde50)
    at /home/research/Downloads/graspit3/src/graspitCore.cpp:286
#9  0x000000000040231d in main (argc=1, argv=0x7fffffffdfe8)
    at /home/research/Downloads/graspit3/src/main.cpp:110

```

jvarley, 2017-08-25:
graspit_commander is a ROS package, so you would need to build it as part of a ros workspace.