graspit-simulator/graspit #102: Writing my own planner?

ilaplace, 2017-03-17:
Hello there,
I want to create my own planner with reineforcement learning, for that I want to use the whole joint space that has been reduced by eigengrasps, could you tell me how to access those states?

jvarley, 2017-03-19:
Just don't use the eigen grasp space.  If whatever gripper you are using doesn't have an eigen.xml similar to:

https://github.com/graspit-simulator/graspit/tree/master/models/robots/Barrett/eigen

Then it will use the full dof space of the hand by default.

ilaplace, 2017-03-21:
Then can I adjust parameters of hand like individual motor positions, hand orientations with corresponding qualitiy metrics?

jvarley, 2017-03-21:
I would suggest either one of these ways forward:
1) write a new energy function for the Simulated Annealing planner built into graspit.  You would be creating the function that rates any given hand - object configuration.  The sim ann code will then find low energy hand-object configurations according to the energy function you have defined.  You will need to create a class following contactEnergy.h/.cpp as an example.  This class computes an energy for a given hand-object state based on how well aligned and how close the hand is to the object, closer and better aligned = lower energy.

https://github.com/graspit-simulator/graspit/blob/master/src/EGPlanner/energy/contactEnergy.cpp#L10

All that really needs to be defined is the energy() function where the energy should be determined based on mHand, mObject.  You will then need to register your new energy function via:

https://github.com/graspit-simulator/graspit/blob/master/src/EGPlanner/energy/searchEnergyFactory.cpp#L40

And you will be able to use it in the EGPlanner. Run graspit, load a world, Grasp->EGPlanner and select your new energy

2) RL + Graspit 
you do not want to use any of the Simulated Annealing infrastructure.  I would recommend doing this in python quite honesty.  Look at:

https://github.com/graspit-simulator/graspit_commander

I had started to do something like this in:
https://github.com/GraspControl/grasp_controller/blob/master/grasp_controller.py
although unfortunately it isn't exactly working at the moment and I have not had time for it.  

But basically, you could do something like:
```
from graspit_commander import GraspitCommander

import time

if __name__ == "__main__":
    GraspitCommander.clearWorld()
    GraspitCommander.loadWorld("plannerMug")
    
    while not GraspitCommander.dynamicAutoGraspComplete():
        r = GraspitCommander.getRobot(0)
        #This is your observation of the current environement
        # you will probably want to augment this with other inputs tactile, rgb, etc
        obs =  r.robot.dofs
        action = take_action(obs)
        GraspitCommander.setRobotDOFForces(action.dofs)
        GraspitCommander.stepDynamics(1)
```

You can use graspit_commander to take screenshots of the graspit environment, get the current robot pose, dof values, joint state, and set the robot pose, dof values.

jvarley, 2017-03-21:
If I hadn't read your first question correctly and you just want to convert a set of eigengrasp amplitudes into the corresponding dof values then look at:
https://github.com/graspit-simulator/graspit/blob/master/include/eigenGrasp.h#L228

ilaplace, 2017-04-04:
Thank you very much!

closed 2017-04-04 by ilaplace