graspit-simulator/graspit #103: Simulation start errors

ilaplace, 2017-03-28:
Hi there,
When I start simulation in dlr_flask world, I'm getting the error:
```
Delta failsafe due to joint
Timestep failsafe reached.

``` 

And If I try barretGlassDyn, when I load
```
Collision found when looking for contacts (for couple of times)
```
And when start simulation:

```
Delta failsafe due to collision: table and _chain1_link2
Collision found when looking for contacts
Timestep failsafe reached.
```
If I try to avoid contact by adjusting puma arm before simulation then then I get:
```
LCP COULD NOT BE SOLVED!
LCP could not be solved.

```
Which is the same one I get after starting the simulation in plannerMug

And after each error, I can't load any other worlds so I have got to close Graspit

mephias, 2017-03-29:
Same thing was happening to me on Ubuntu 16.04. Only solution I could figure out for now was reinstalling OS to Debian (tho most likely thanks to my inexperience with debian, im still getting different errors but not "LCP could not be solved ones", and I saw it somehow working on different machine with debian). My guess is that external libraries/packages got updated and graspit isnt handling that very well (debian gets updates slower than ubuntu). So if you did package update recently, I would say try to reinstall them (who knows) or somehow install older ones and if it wont help then either try debian (if it isnt that much of a problem) or wait till someone with better understanding of how graspit works will reply but I still havent got answer when I opened my issue about "LCP could not be solved".

ilaplace, 2017-04-20:
Installing debian crushed my system fully :P

closed 2017-07-28 by mateiciocarlie