graspit-simulator/graspit #105: graspit_simulator.exe has stopped working

TirawatT, 2017-03-29:
[image]

Error , when i'm run graspit_simulator.exe 

I'm build graspit 3.0 and follow http://graspit-simulator.github.io/build/html/index.html 
and Run program on machine windows 7 64

what's wrong ?

Thank you so much.

jvarley, 2017-03-30:
I have very little experience running GraspIt! on windows unfortunately.   If you are able to get a stack trace I might be able to help a bit more.

@JenniferBuehler  would probably be more helpful, but I would image she would need more information as well.

JenniferBuehler, 2017-04-01:
Unfortunately I won't be of much help either, I don't work on Windows. I just make it compile with MinGW for the PR about the move from qmake to CMake... I also tried running it back then and everything worked, but that was about a year ago (?) on an older version of the source...

TirawatT, 2017-04-04:
Thank you so much.

TirawatT, 2017-04-04:
Thank you so much.

closed 2017-04-04 by TirawatT