graspit-simulator/graspit #106: Bullet Dynamics Can't Start

ilaplace, 2017-03-29:
Since I couldn’t resolve my issue: 
[https://github.com/graspit-simulator/graspit/issues/103](url)

I wanted to use bullet dynamics since I need gravity. I followed the tutorial [http://graspit-simulator.github.io/build/html/dynamics.html](url). But it seems like it doesn't work, I keep getting the same errors.

jvarley, 2017-03-30:
I have been a bit busy,  but will look over the dynamics issues soon.  

You should definitely not be getting the same error with Bullet though as it does not use the LCP solver inside GraspIt!.  You may want to rm -rf your build directory, and change the CMakeLists.txt to use bullet 
specifically change this line:

set(DYNAMICS_ENGINE GRASPIT_DYNAMICS CACHE PATH "back end for GraspIt! Dynamics, current options are GRASPIT_DYNAMICS, BULLET_DYNAMICS")

to 

set(DYNAMICS_ENGINE BULLET_DYNAMICS CACHE PATH "back end for GraspIt! Dynamics, current options are GRASPIT_DYNAMICS, BULLET_DYNAMICS")

and then rerun:
```
mkdir build
cd build
cmake ..
make -j5
```

We also made a rather large change to GraspIt!, removing its older custom math library and replacing most of the functionality with Eigen.  It maybe worth checking out an older commit of GraspIt! prior to the switch to Eigen and seeing if that it works as planned.  Maybe give this one a try:
https://github.com/GraspControl/graspit/tree/aeb9972a56d8e7326d639138e24bc9e645fd815f

jvarley, 2017-03-30:
Does the dynamics engine run for you on the plannerMug.xml world, using either bullet or graspit dynamics?

jvarley, 2017-03-30:
You can add gravity in Bullet using:
https://github.com/graspit-simulator/graspit/blob/master/src/dynamics/bulletDynamics.cpp#L67

ilaplace, 2017-03-30:
After I've created build folder under graspit_source, I'm not getting LCP errors, When first start it asks me this: 
[image]

I could not run dynamic engine in any world files, Currently It seems like when dynamic is somehow on, in the planner mug, although after I've made auto-grasp it didn't made any contact:
[image]

In barretGlassDyn things go pretty weird:
[image]

Dlr world goes crazy too: 
[image]

jvarley, 2017-04-01:
Can you try followign the readme here:
https://github.com/GraspControl/grasp_control_ws

This was how I was running dynamics inside graspit several months ago.  It should work fine without bullet at least for the barrett hand with the mug.  Take a peek at 

https://github.com/GraspControl/grasp_controller/blob/master/grasp_controller.py

before you run it to get a feel for what it does.  

The dynamics engine doesn't always work perfectly  in terms of mirroring what would happen in the real world, but what you are posting does not look like normal behavior.

ilaplace, 2017-04-01:
It works exactly the same although I did something tiny different:
'wstool update' didn't work for me since I have an SSH issue of something, so I manually downloaded graspit-ros under the GraspControl repo, maybe I got the wrong version of it?

By the another thing I did different was to configure Bullet Dynamics with editing the:`
 /home/ilaplace/ctrl_grasp_ws/src/graspit-ros/graspit/graspit_source/CMakeCache.txt`

jvarley, 2017-04-03:
If you follow the Readme from:
https://github.com/GraspControl/grasp_control_ws

Without Bullet it still doesn't work?

ilaplace, 2017-04-03:
Nothing seems different, same bullet engine issues, and the same grasp dynamic ones

jvarley, 2017-04-03:
So when you run grasp_controller.py you don't end up with the hand in a reasonable configuration around the mug?

ilaplace, 2017-04-03:
No, it works fine, but when start simulation thing go crazy

jvarley, 2017-04-04:
the rqt errors are because the fork of graspit_interface in graspcontrol has additional services related to controlling a robot in dynamics mode (SetRobotDOFForces, StepDynamics, ToggleDynamicsController) that do not exist in the main graspit_interface repo at this time.

closed 2017-07-28 by mateiciocarlie