graspit-simulator/graspit #110: GraspIt/moveit_simple_grasps

gecastro, 2017-05-16:
I'm looking at using GraspIt with ROS and I was wondering if there is any method to give a pre-computed grasp and refine it in GraspIt.

I'm looking at the option of pipelining moveit_simple_grasps (or simply enter a manual input) to pre-generate solutions and using GraspIt to refine them and compute quality.

Is there any work around this that I could use?

jvarley, 2017-05-16:
Depends on how good the precomputed grasp is.  
Option 1): 
Adjust how open/close hand is and distance along approach direction
Use graspit_commander
```
def plan_list_grasps(meshfile):
    gc = graspit_commander.GraspitCommander()
    gc.importGraspableBody(meshfile)
    gc.importRobot("BarrettBH8_280")
                                                                                                                       
    pre_grasps = GridSampleClient.getSamples(10).grasps
                                                                                        
    completion_results = []
    for i, pre_grasp in enumerate(pre_grasps):
        gc.toggleAllCollisions(False)
        gc.forceRobotDof([0,0,0,0])
        gc.setRobotPose(pre_grasp.pose)                                                                                                                    
        gc.toggleAllCollisions(True)
        gc.findInitialContact()                                                                                                                                                                                                                                                                                                                                        
        gc.autoGrasp()                                                                                                                  
        volume_quality = gc.computeQuality().volume
        result = gc.getRobot(0)
        grasp = copy.deepcopy(pre_grasp)
        grasp.pose = gc.getRobot(0).robot.pose
        grasp.volume_quality = volume_quality
        completion_results.append((volume_quality, grasp))

    completion_results.sort()
    completion_results.reverse()

    grasps = []
    for quality, grasp, robot_state in completion_results:
        if quality > 0:
            grasps.append((quality,grasp,robot_state))

   return grasps
```

Option 2) Want entirely new grasp seeded with pre grasp.  Look at Graspit's online planner.  It will start the simulated annealing planner near a prespecified seed location.  It is not currently exposed via graspit_commander though.  Although it would not be too hard to do.  You can play with it inside graspit.

jvarley, 2017-05-16:
[image]

closed 2017-09-20 by jvarley