graspit-simulator/graspit #112: Load grasps failed

JeremySI, 2017-06-14:
I do follow the instruction on the usermaul and try to connnect to the database. The connection is successed and I can load the model in to it.However, when I try to load the grasps , the terminal respond that the load grasps failed. What I import is the humanhand and the model in the database. 
Thank you for answering my question.

jvarley, 2017-06-15:
perhaps a similar problem to:
https://github.com/graspit-simulator/graspit/pull/26

JeremySI, 2017-06-15:
Thanks for reply, I tried to load grasps using Barrett hand, it works fine with Barrett hand, but when I use with Humanhand, the console shows load grasps failed. It should work with the 3 different human hand (16 DOF,20DOF,Altaxe)as well right?

jvarley, 2017-06-15:
Make sure your DB has hand entries for the human hand you are trying to load:

Look at the DBName attribute in the xml file.  
https://github.com/graspit-simulator/graspit/blob/master/models/robots/HumanHand/HumanHand16DOF.xml#L2

Sorry I am not much help on this, I have not used the CGDB with the human hand, I have only used it with the Barrett. I am not exactly sure how many grasps are in there for the human hand.

JeremySI, 2017-06-16:
Thank you for helping me！

closed 2017-09-20 by jvarley