graspit-simulator/graspit #125: How can we make not display graspable object coordinates?

knorth55, 2017-08-10:
I'm planning to collect RGB image of objects, but it display graspable object coordinates.
How can we make this coordinates not display in captured image?
I want to remove this yellow x-y-z coordinates from the image.
[image]

jvarley, 2017-08-10:
[image]

[image]

[image]

Element->Body Properties  then toggle Show Axes. 

[image]

[image]

If you look in Body.cpp, you can find the body constructor and set showAx variable to false, you may also want to flop off showFC (friction cones) and showVC (virtual contacts. )

knorth55, 2017-08-10:
thank you so much!
I'm using GraspIt! from graspit_commander, but it seems I cannot do this process via moveit_interface.
Anyway, thank you so much!

jvarley, 2017-08-10:
You could add functionality to do this to graspit_interface/graspit_commander and make a PR :)

knorth55, 2017-08-10:
Let me take a look for it!

closed 2017-08-10 by knorth55