graspit-simulator/graspit #126: RobotiQ 2Finger Issue

sameergrey, 2017-08-23:
Hi,

I am using Robotiq 2 Finger Gripper for Grasp Planning using Graspit.But whenever i tried to do EigenGrasp Planner the result is not proper My Robot Hand is always in some degree inclination.
I am using simple cube to grasp.
Why is it Happening i am attaching the Model File of My Robot hand as well as some Planning Pictures.

I am using Sim Annaling Grasp Planner and Hand Contacts as a Energy Formulation.This all are in Graspit simulator.

I am also attaching URDF of my robot. 
Kindly Help Me on this.
[image]
[image]
[robotiq_arg2f_140_model.zip](https://github.com/graspit-simulator/graspit/files/1245150/robotiq_arg2f_140_model.zip)

jvarley, 2017-08-23:
Hi,

You are running the Hand Contacts energy formulation.  It minimizes the total distance between the virtual contacts (red arrows) and your hand.  The picture you attached looks like it is well minimized.  If you want the object to stay between the tips of your gripper, perhaps removed some of the lower virtual contacts.  

I would recommend running the Guided Potential Quality, it uses the Hand Contacts energy to pull the hand close to the object, and then the energy measures whether or not the grasp exhibits force closure.  We don't have an energy function that ensure than hand stays level in a world frame, you can play with the search space and not allow it to try hand configurations other than with fixed roll pitch or yaw values.  

Also, would you be willing to make a PR to graspit with your hand model?

-Jake

sameergrey, 2017-08-24:
Hi,
Thanks a lot for your reply.
Yes i used the  Guided Potential Quality Energy Formulation but still the results are poor.
I done one experiment with the exsting Robot Model of Graspit that is Micro Gripper when i plan with that Robot hand it seems it gives me the proper Grasping pose which i intend to do with my gripper.

Where i am missing anything? Please let me help on that i am attaching two image one with my hand and one is with the exsting model.

Yes i play around with the Space Search also i tried Axis-Angle and other options but the results are poorer.

I have question how does the Graspit Able to know this should be my contact points to grasp because my hand always in the tilt mode i don't why?

Yes Once all are working fine i will post my model to the Repo.
[image]
[image]

jvarley, 2017-08-24:
do you have any of the meshes for the hand as obj, stl, or ply? Open them in meshlab and view the surface normals, verify that the extend out of the mesh rather than inwards.  I am thinking maybe the black meshes on your hand have their surface/vertex normals facing the wrong way.

jvarley, 2017-08-28:
any luck?

sameergrey, 2017-08-29:
Sorry for late response.Nope...I tried inverting normals of my stl and then created graspit model but then contact points on my model's finger tips started pointing outward as shown in screenshot.I assume normals of my mesh are correctly pointing outwards.My primary observation after planning with this gripper is that the contact point normals on both of my finger tips are always normal to each other and these finger joints are not getting closed for all the poses and just hand pose transfers with same joint angle for all the poses.Please suggest some other way to tackle this problem.Also is prob because of mimic joints present in urdf of robotiq grippers?I made them revolute while converting but the actual mechanism of this hand which works in ROS has mimic joints.Even after making them revolute no luck 
[image]

jvarley, 2017-08-29:
Can you send me the stls

sameergrey, 2017-08-29:
I have attached meshes which I am using.These are the same meshes on ROS-Industrial github repository with branch jade-devel
 https://github.com/ros-industrial/robotiq/tree/jade-devel/robotiq_arg2f_model_visualization
[meshes_robotiq_140.zip](https://github.com/graspit-simulator/graspit/files/1260340/meshes_robotiq_140.zip)

jvarley, 2017-08-29:
This image shows running our planner with a single virtual contact with a single small object

[image]

We are using the contact energy here:
https://github.com/graspit-simulator/graspit/blob/master/src/EGPlanner/energy/contactEnergy.cpp#L45

This energy minimizes two things:
1) distance between virtual contact and object, this appears to work fine
2) cn = contact->getWorldNormal();
    n = p.normalized();
double d = 1 - cn.dot(n);

keeping the hand well aligned with the object.  This part is not working.  Either there is something wrong with your mesh, or how you defined your virtual contacts.

Look at:
https://github.com/graspit-simulator/graspit/blob/master/models/robots/Barrett/virtual/all_16_contacts.xml

We have updated the syntax for defining virtual contacts, I would port your current virtual contacts file to match this, as it makes it easier to look at, and then I would pay close attention to how you define the virtual contact normal.

https://github.com/graspit-simulator/graspit/blob/master/models/robots/Barrett/virtual/all_16_contacts.xml#L37

I looked at your meshes in meshlab, and they seem ok, I think the normals for your virtual contacts are defined incorrectly.

jvarley, 2017-08-29:
You can try this for your contacts.vgr, and it fixes the problem.

robotiq_arg2f_140_model
2
1 2
20
1.01126e-16 0 0 0 0 1.65151 
0.855408 0 0 0 0 0.855408 
0.264335 0.813541 0 0 0 0.855408 
-0.692039 0.502796 0 0 0 0.855408 
-0.692039 -0.502796 0 0 0 0.855408 
0.264335 -0.813541 0 0 0 0.855408 
1 0 0 0 0 0 
0.707107 0.707107 0 0 0 0 
6.12323e-17 1 0 0 0 0 
-0.707107 0.707107 0 0 0 0 
-1 1.22465e-16 0 0 0 0 
-0.707107 -0.707107 0 0 0 0 
-1.83697e-16 -1 0 0 0 0 
0.707107 -0.707107 0 0 0 0 
0.855408 0 0 0 0 -0.855408 
0.264335 0.813541 0 0 0 -0.855408 
-0.692039 0.502796 0 0 0 -0.855408 
-0.692039 -0.502796 0 0 0 -0.855408 
0.264335 -0.813541 0 0 0 -0.855408 
1.01126e-16 0 0 0 0 -1.65151 
11.5314 -30.6451 1.60615e-05
0.39287 0.587923 0.39287 -0.587923
11.5314 -30.6451 1.60615e-05
-0.382615 -0.923908 7.37451e-08
0.3
0 2
20
1.01126e-16 0 0 0 0 1.65151 
0.855408 0 0 0 0 0.855408 
0.264335 0.813541 0 0 0 0.855408 
-0.692039 0.502796 0 0 0 0.855408 
-0.692039 -0.502796 0 0 0 0.855408 
0.264335 -0.813541 0 0 0 0.855408 
1 0 0 0 0 0 
0.707107 0.707107 0 0 0 0 
6.12323e-17 1 0 0 0 0 
-0.707107 0.707107 0 0 0 0 
-1 1.22465e-16 0 0 0 0 
-0.707107 -0.707107 0 0 0 0 
-1.83697e-16 -1 0 0 0 0 
0.707107 -0.707107 0 0 0 0 
0.855408 0 0 0 0 -0.855408 
0.264335 0.813541 0 0 0 -0.855408 
-0.692039 0.502796 0 0 0 -0.855408 
-0.692039 -0.502796 0 0 0 -0.855408 
0.264335 -0.813541 0 0 0 -0.855408 
1.01126e-16 0 0 0 0 -1.65151 
11.0578 -30.449 -8.03263e-06
0.39287 0.587923 0.39287 -0.587923
11.0578 -30.449 -8.03263e-06
-0.382615 -0.923908 7.37452e-08
0.3

sameergrey, 2017-08-30:
Thanks alot for the help.The model worked after adding above data in contacts.vgr file.It solved my problem partially.But now my concern is the hand  remains open even if pose is properly aligned to object.So why cant I define contact point on other finger as well so that hand will get closed because in our project we need to deal with cluttered environment where object to be grasped might be very very close to obstacles and in that case planner might fail since hand which is open as of in current situation and might not find any solution since it might collide with environment obstacles.Also I am unaware of new way to define virtual contact point since the version which I am using defines contact points through interactive visualizer and creates only vgr file. I am using graspit ROS interface provided over here
https://github.com/JenniferBuehler/graspit-pkgs which uses graspit version 2.2.0 forked which is compatible with graspit. So I might not be able to access latest changes in graspit.
 Kindly have a look at current situation of my model in my custom made environment
[image]

jvarley, 2017-08-30:
add sphere , add hand, 

menu -> grasp -> virtual contacts
[image]

bump the sphere into the hand, and click mark contact.  Do this several times, then click save in the virtual grasp window. 

robotiq_arg2f_140_model
4
1 2
20
1.01126e-16 0 0 0 0 1.65151 
0.855408 0 0 0 0 0.855408 
0.264335 0.813541 0 0 0 0.855408 
-0.692039 0.502796 0 0 0 0.855408 
-0.692039 -0.502796 0 0 0 0.855408 
0.264335 -0.813541 0 0 0 0.855408 
1 0 0 0 0 0 
0.707107 0.707107 0 0 0 0 
6.12323e-17 1 0 0 0 0 
-0.707107 0.707107 0 0 0 0 
-1 1.22465e-16 0 0 0 0 
-0.707107 -0.707107 0 0 0 0 
-1.83697e-16 -1 0 0 0 0 
0.707107 -0.707107 0 0 0 0 
0.855408 0 0 0 0 -0.855408 
0.264335 0.813541 0 0 0 -0.855408 
-0.692039 0.502796 0 0 0 -0.855408 
-0.692039 -0.502796 0 0 0 -0.855408 
0.264335 -0.813541 0 0 0 -0.855408 
1.01126e-16 0 0 0 0 -1.65151 
5.7779 -28.2624 -2.23391e-09
0.39287 0.587923 0.39287 -0.587923
5.7779 -28.2624 -2.23391e-09
-0.382615 -0.923908 7.37452e-08
0.3
1 2
20
1.01126e-16 0 0 0 0 1.65151 
0.855408 0 0 0 0 0.855408 
0.264335 0.813541 0 0 0 0.855408 
-0.692039 0.502796 0 0 0 0.855408 
-0.692039 -0.502796 0 0 0 0.855408 
0.264335 -0.813541 0 0 0 0.855408 
1 0 0 0 0 0 
0.707107 0.707107 0 0 0 0 
6.12323e-17 1 0 0 0 0 
-0.707107 0.707107 0 0 0 0 
-1 1.22465e-16 0 0 0 0 
-0.707107 -0.707107 0 0 0 0 
-1.83697e-16 -1 0 0 0 0 
0.707107 -0.707107 0 0 0 0 
0.855408 0 0 0 0 -0.855408 
0.264335 0.813541 0 0 0 -0.855408 
-0.692039 0.502796 0 0 0 -0.855408 
-0.692039 -0.502796 0 0 0 -0.855408 
0.264335 -0.813541 0 0 0 -0.855408 
1.01126e-16 0 0 0 0 -1.65151 
22.6106 -35.2333 -8.46434e-12
0.39287 0.587923 0.39287 -0.587923
22.6106 -35.2333 -8.46434e-12
-0.382615 -0.923908 -1.52725e-13
0.3
0 2
20
1.01126e-16 0 0 0 0 1.65151 
0.855408 0 0 0 0 0.855408 
0.264335 0.813541 0 0 0 0.855408 
-0.692039 0.502796 0 0 0 0.855408 
-0.692039 -0.502796 0 0 0 0.855408 
0.264335 -0.813541 0 0 0 0.855408 
1 0 0 0 0 0 
0.707107 0.707107 0 0 0 0 
6.12323e-17 1 0 0 0 0 
-0.707107 0.707107 0 0 0 0 
-1 1.22465e-16 0 0 0 0 
-0.707107 -0.707107 0 0 0 0 
-1.83697e-16 -1 0 0 0 0 
0.707107 -0.707107 0 0 0 0 
0.855408 0 0 0 0 -0.855408 
0.264335 0.813541 0 0 0 -0.855408 
-0.692039 0.502796 0 0 0 -0.855408 
-0.692039 -0.502796 0 0 0 -0.855408 
0.264335 -0.813541 0 0 0 -0.855408 
1.01126e-16 0 0 0 0 -1.65151 
3.61796 -27.3679 8.02741e-06
0.39287 0.587923 0.39287 -0.587923
3.61796 -27.3679 8.02741e-06
-0.382615 -0.923908 7.37452e-08
0.3
0 2
20
1.01126e-16 0 0 0 0 1.65151 
0.855408 0 0 0 0 0.855408 
0.264335 0.813541 0 0 0 0.855408 
-0.692039 0.502796 0 0 0 0.855408 
-0.692039 -0.502796 0 0 0 0.855408 
0.264335 -0.813541 0 0 0 0.855408 
1 0 0 0 0 0 
0.707107 0.707107 0 0 0 0 
6.12323e-17 1 0 0 0 0 
-0.707107 0.707107 0 0 0 0 
-1 1.22465e-16 0 0 0 0 
-0.707107 -0.707107 0 0 0 0 
-1.83697e-16 -1 0 0 0 0 
0.707107 -0.707107 0 0 0 0 
0.855408 0 0 0 0 -0.855408 
0.264335 0.813541 0 0 0 -0.855408 
-0.692039 0.502796 0 0 0 -0.855408 
-0.692039 -0.502796 0 0 0 -0.855408 
0.264335 -0.813541 0 0 0 -0.855408 
1.01126e-16 0 0 0 0 -1.65151 
24.9196 -36.1895 1.4807e-05
0.39287 0.587923 0.39287 -0.587923
24.9196 -36.1895 1.4807e-05
-0.382615 -0.923908 1.29566e-13
0.3

I don't know if Jennifer is continuing to develop those packages.  We are actively maintaining:
https://github.com/graspit-simulator/graspit_interface

It does require ros though...

sameergrey, 2017-08-31:
@jvarley  Thanks alot for your help.Highly appreciated.Actually I was not assigning any virtual contacts by procedure stated by you.Now by problem is solved completely.Thanks once again. :)
We can close this issue now.If possible include this Robotiq model in your robot models directory so that other people can also get it for use.

jvarley, 2017-08-31:
Would you mind making a pull request with the hand?

sameergrey, 2017-09-01:
Sure will make it in this week.

majiqiang, 2019-05-21:
hello，i want to insult you about gripper model in graspit based on ubuntu16.04 not ros. Could you tell me how to build its model's xml file because i want to import ourselves gripper into graspit. Our hand is similar to the robotiq, so i want to use our hand in graspit! Thank you!
[image]
This is our gripper.

jih189, 2019-06-26:
Hello. For gripper of baxter, i think it has the similar problem because the  energy is always positive even it has sampled a pose where it can grasp the object. Please help to fix it.

jih189, 2019-06-27:
I think it can be fixed easily. All you need to do is changing the normal in the .graspit/models/robots/baxter/virtual/contacts.xml. You can see the normal are wrong except the first one.