graspit-simulator/graspit #139: jaco_robot model broken

mattcorsaro1, 2018-02-08:
One of the fingers of the jaco_robot appears to be broken. Nothing happens after I send an open command over graspit_interface. I have a fresh install of the latest version of graspit. Here's the terminal output when I import the hand:

```
Creating base...

Setting up 3 degrees of freedom...
Creating 3 kinematic chains (fingers)...
Chain 0:   Creating joints
   Joint 0
  Creating links
   Link 0
  Creating dynamic joints
Chain 1:   Creating joints
   Joint 0
  Creating links
   Link 0
  Creating dynamic joints
Chain 2:   Creating joints
   Joint 0
  Creating links
   Link 0
  Creating dynamic joints
No normalization data found; using factors of 1.0
Read 1 eigengrasps from EG file
Using eigengrasps from file: eigen/eigen.xml
Loaded virtual contacts from file virtual/contacts.xml
```

jvarley, 2018-02-08:
Make this:
https://github.com/graspit-simulator/graspit/blob/master/models/robots/jaco_robot/jaco_robot.xml#L50
d1+210  ----> d1+0
That will get the hand to show up correctly.  

That hand does not play well with the dynamics engine as is.  I am not sure why.  In the jaco_robot.xml file try playing around with different values for <Kd>1e+07</Kd> and <maxEffort>-1.5e+5</maxEffort> maybe look at what has been used for the Barrett hand.

JenniferBuehler, 2018-02-12:
After some updates to the graspit code, now there is a new version of the Jaco hand. The model has been generated with [urdf2graspit](https://github.com/JenniferBuehler/graspit-pkgs/wiki/urdf2graspit) from the [jaco-arm-pkgs](https://github.com/JenniferBuehler/jaco-arm-pkgs/wiki) before it was added to this repo. This has been a while ago, so it could probably use an update. I will try to put a PR together soon. Meanwhile you could try to use the [jaco_graspit_sample model here](https://github.com/JenniferBuehler/graspit-pkgs/tree/master/jaco_graspit_sample), does it work with that one?