graspit-simulator/graspit #141: Using graspit! for closed chain fingers

echeng22, 2018-03-06:
Hi, I've been trying to use graspit for the robotiq 2 finger 85 model and I've been running into some issues on trying to get it to move. Below is an image of the gripper I'm working with. When I use the interaction mode mouse to try to move the joints when collision is ON, I receive a stream of "Robot joint interpolation errors".  When collision is off, the top left and right joints can move without moving the rest of the gripper (see image below). 

The way I have the robot described is two kinematic chains for each gripper finger, and I'm guessing that at the initial position, they are already in collision, and therefore run into the errors above when trying to interact with them. Is there a way that I could set up the robot description file to get this to work for the closed chain setup? Or maybe an alternative way to setup the robot description xml?

Thanks!
I've attached the robot model and robot description files I'm using below, in case that helps in seeing the issue.

[image]
[image]

[robotiq_2f_85.tar.gz](https://github.com/graspit-simulator/graspit/files/1783198/robotiq_2f_85.tar.gz)

majiqiang, 2019-05-21:
hello，i want to insult you about gripper model in graspit based on ubuntu16.04 not ros. Could you tell me how to build its model's xml file because i want to import ourselves gripper into graspit. Our hand is similar to the robotiq, so i want to use our hand in graspit! thank you!
[image]

majiqiang, 2019-05-21:
sorry to bother you, i have seen your robot description files, i really want to know how you build these files,such as eigen, iv,and robotiq_2f_85.xml.

majiqiang, 2019-06-12:
Hi ! have you settled it ? also I have a gripper similar to you .Right now i just have the stl file. Could you help me obtain the robot description files? Thank you very much!