graspit-simulator/graspit #150: Make failed with ivmgr.cpp

Tortes, 2019-01-22:
I'm using Ubuntu16.04 and following the guide. The error occurs when I finished cmake and began to make. The error message is as below
```
[ 41%] Building CXX object CMakeFiles/graspit.dir/src/ivmgr.cpp.o
/home/tortes/Desktop/graspit/src/ivmgr.cpp: In member function ‘void IVmgr::transRot(DraggerInfo*)’:
/home/tortes/Desktop/graspit/src/ivmgr.cpp:673:60: error: no match for ‘operator[]’ (operand types are ‘const SbRotation’ and ‘int’)
     desiredRotation.x() = myCenterball->rotation.getValue()[0];
                                                            ^
/home/tortes/Desktop/graspit/src/ivmgr.cpp:674:60: error: no match for ‘operator[]’ (operand types are ‘const SbRotation’ and ‘int’)
     desiredRotation.y() = myCenterball->rotation.getValue()[1];
                                                            ^
/home/tortes/Desktop/graspit/src/ivmgr.cpp:675:60: error: no match for ‘operator[]’ (operand types are ‘const SbRotation’ and ‘int’)
     desiredRotation.z() = myCenterball->rotation.getValue()[2];
                                                            ^
/home/tortes/Desktop/graspit/src/ivmgr.cpp:676:60: error: no match for ‘operator[]’ (operand types are ‘const SbRotation’ and ‘int’)
     desiredRotation.w() = myCenterball->rotation.getValue()[3];
                                                            ^
CMakeFiles/graspit.dir/build.make:1243: recipe for target 'CMakeFiles/graspit.dir/src/ivmgr.cpp.o' failed
make[2]: *** [CMakeFiles/graspit.dir/src/ivmgr.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/graspit.dir/all' failed
make[1]: *** [CMakeFiles/graspit.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

```
Thanks for any help.

jsll, 2019-02-09:
This is by far not an ideal solution but I solved it by simply commenting out the lines corresponding to those problems which are lines 673-676 in src/ivmgr.cpp and everything worked fine from there on. If there is a neater solution to this in the future you can simply reinstall graspit then.

Tortes, 2019-02-14:
> This is by far not an ideal solution but I solved it by simply commenting out the lines corresponding to those problems which are lines 673-676 in src/ivmgr.cpp and everything worked fine from there on. If there is a neater solution to this in the future you can simply reinstall graspit then.

Thanks! It works well!

closed 2019-02-14 by Tortes