graspit-simulator/graspit #152: Dynamics engines not working properly in GraspIt

fbottarel, 2019-04-19:
Hello, 
I have downloaded and compiled successfully the simulator. I can load any example and run any tutorial __except__ the [one regarding dynamics](https://graspit-simulator.github.io/build/html/getting_started.html#dynamics-example) (worlds/barrettGlassDyn.xml). The scene loads up, but it appears like the hand is already in contact with the table, therefore the `Auto-Grasp` doesn't do anything. I get the following console output: 
```
Using default inertia matrix
Using default center of gravity
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
Collision found when looking for contacts
```

Moreover, if I try to start the dynamics simulation nothing happens in the GUI, and the time counter does not move from 00.00. Nothing moves in the scene. I only get a `Timestep failsafe reached.` warning as console output. However, pausing the simulation brings up another warning

```
Coin warning in SbRotation::setValue(): rotateTo argument has zero length.
Coin warning in SbRotation::setValue(): rotateTo argument has zero length.
Delta failsafe due to collision: table and _chain1_link2
```
Unpausing the simulation doesn't fix things. I compiled the simulator with both the graspit dynamics engine and Bullet, none work (using bullet makes the sim crash, but I think I haven't compiled it corectly). These errors consistently showed up for me on 3 different machines, running Ubuntu 16.04, 14.04 and 18.04.

I wonder if anyone has any clue about what is going on here.

wutter, 2019-04-30:
Same problem for me, collision is between finger-table, after modifing world file I got the hand to open and close. But glass was not affected by gravity, thus did not start rolling as suggested in docs. Ubuntu 18.04. Currently trying to get it running on Windows 7 and see what happens there.

fbottarel, 2019-04-30:
@wutter have you tried with bullet as well?

wutter, 2019-05-02:
Cool, that works! Now only problem is that the whole arm just goes bananas 😄, using bullet tag 2.88, meby older version would do better.

fbottarel, 2019-05-03:
@wutter let me know if you make any advancement. I am just about to give up on graspit

wutter, 2019-05-07:
Found out you can get Bullet without compiling, on clean install of ubuntu 18.04 this should get it all prepared:
sudo apt-get install make gcc libqt4-dev libqt4-opengl-dev libqt4-sql-psql  libcoin80-dev libsoqt4-dev libblas-dev  liblapack-dev  libqhull-dev  libeigen3-dev libcanberra-gtk-module libcanberra-gtk3-module cmake-curses-gui git 
libbullet-dev libbullet-extras-dev libbullet2.87 libbullet-extras2.87 libbulletml0v5 libbulletml-dev 
libglu1-mesa-dev freeglut3-dev mesa-common-dev

with this setup I do not get memory leak crashes, but multiple robot-bodies connected still seem to be broken. When using it with only one robot, it seems kinda OK, until collision occurs, as when that happens, GraspIt! gets stuck in a loop saying "Collision found when looking for contacts".