graspit-simulator/graspit #155: Schunk WSG 50

mattcorsaro1, 2019-05-21:
Hi,

I'd like to use the Schunk WSG 50 hand. I generated a GraspIt model from [wsg50-ros-pkg](https://github.com/nalt/wsg50-ros-pkg)'s xacro using urdf2graspit. The model is in my fork [here](https://github.com/mattcorsaro1/graspit/tree/wsg_50). When I run the openclose plugin with it, nothing happens and I get this error:

> Robot joint interpolation error
> Interpolation failed!

Any idea what the problem could be? A discussion about the EigenGrasp file is [here](https://github.com/JenniferBuehler/graspit-pkgs/issues/47).

Thank you!
Matt

majiqiang, 2019-05-22:
Can GRASPIT also use urdf files? how  do you do it? my gripper is similar to robotiq, i'm very confused now!

mattcorsaro1, 2019-05-22:
You'll need to convert your URDF to a GraspIt robot model using [urdf2graspit](https://github.com/JenniferBuehler/graspit-pkgs/wiki/urdf2graspit)

majiqiang, 2019-05-23:
> You'll need to convert your URDF to a GraspIt robot model using [urdf2graspit](https://github.com/JenniferBuehler/graspit-pkgs/wiki/urdf2graspit)
OK，thank you very much .What's more, how do you get urdf model? i just know a tool , SolidWorks plugin, maybe convert stl or other 3D model to urdf. I don't know if it's right, so i want to ask for your help.

mattcorsaro1, 2019-05-23:
Yeah, if you have an stl file, you should be able to generate a URDF with [this plugin](http://wiki.ros.org/sw_urdf_exporter).

majiqiang, 2019-05-24:
> Yeah, if you have an stl file, you should be able to generate a URDF with [this plugin](http://wiki.ros.org/sw_urdf_exporter).

yeah！ i know this plugin, but i am confused that how to use this plugin, my gripper is complex, it has a parallelogram mechanism link, i don't know how to bulid it. Do you have a good method?

mattcorsaro1, 2019-05-29:
I edited the defaultVelocity and joint limits in my model and it seems to work now. I'll submit a PR once I've tested it.

closed 2019-05-29 by mattcorsaro1

majiqiang, 2019-05-30:
> I edited the defaultVelocity and joint limits in my model and it seems to work now. I'll submit a PR once I've tested it.

Hello，Matt！Honor to receive your reply. Recently I try to export my gripper to urdf file firstly, but i failed. It  looks very irregular in RVIZ. I really don't know how to do, could you help me ?
This is my mesh model with coordinate system and my generated urdf file. And maybe the coordinate is wrong, I don't know.
[image]
[image]
[装配体1.zip](https://github.com/graspit-simulator/graspit/files/3236057/1.zip)
[myurdf.tar.gz](https://github.com/graspit-simulator/graspit/files/3236070/myurdf.tar.gz)

majiqiang, 2019-06-13:
> I edited the defaultVelocity and joint limits in my model and it seems to work now. I'll submit a PR once I've tested it.

Hello, Matt, could you help me generate urdf file or robot description files? I 'm in trouble!

majiqiang, 2019-06-25:
[image]
i met same problem!
how do you solve?

> Robot joint interpolation error
Interpolation failed!