graspit-simulator/graspit #156: how to translate the 3D model stl to  xml and iv?

majiqiang, 2019-05-22:
i use our gripper model in graspit based on ubuntu16.04 not ros. i want to build its model's xml file because i want to import ourselves gripper into graspit. Our hand is similar to the robotiq,  thank you!

mattcorsaro1, 2019-05-22:
You'll need to convert your URDF to a GraspIt robot model using [urdf2graspit](https://github.com/JenniferBuehler/graspit-pkgs/wiki/urdf2graspit)

jih189, 2019-06-26:
you can use blender to convert it. By the way, remember to repose the object so it can convert.

majiqiang, 2019-06-26:
> you can use blender to convert it. By the way, remember to repose the object so it can convert.

what is the blender？

closed 2019-06-26 by majiqiang

jih189, 2019-06-26:
check this out https://www.blender.org/. For the method to convert, you can search with google.