graspit-simulator/graspit #161: Undefined reference to symbol '_ZN7QCursorC1EN2Qt11CursorShapeE'

ItzMeJP, 2019-11-04:
I am trying to compile the graspit from the source. For this I am using Ubuntu18, however i get the follow problem:

> /usr/bin/ld: CMakeFiles/graspit_simulator.dir/src/main.cpp.o: undefined reference to symbol '_ZN7QCursorC1EN2Qt11CursorShapeE'
> //usr/lib/x86_64-linux-gnu/libQtGui.so.4: error adding symbols: DSO missing from command line
> collect2: error: ld returned 1 exit status
> CMakeFiles/graspit_simulator.dir/build.make:107: recipe for target 'graspit_simulator' failed
> make[2]: *** [graspit_simulator] Error 1
> CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/graspit_simulator.dir/all' failed
> make[1]: *** [CMakeFiles/graspit_simulator.dir/all] Error 2
> Makefile:129: recipe for target 'all' failed
> make: *** [all] Error 2

anyone can help me with this issue?

dnandha, 2020-02-18:
Having the same issue!

lepalom, 2020-09-16:
I'm curious how did you build graspit with Qt5 ...

cellolegs, 2021-01-07:
Did anyone find a fix for this? I have encountered the same error. (also interested in how to compile with QT5?!?!)

lepalom, 2021-01-07:
@cellolegs I think that current version of graspit cannot be build using Qt5. It uses a lot of methods from Qt4 and Qt3. 

I did some work creating a version of graspit that it builds with Qt5. It can be found [here](https://github.com/iocroblab/graspit)

 It builds in a Debian 10 Buster and Ubuntu 18.04 with SoQt5 (from [here] (https://launchpad.net/~deb-rob/+archive/ubuntu/ros-bionic)  and Qt5. It Ubuntu Focal should work too.

- You can load a model and it appears.
- Graspit server is disabled. Qt5 have changed too many things (at least in a first view) so is a pending task.
- The main form fails is some actions, but it's just review all the Actions/Slots.
- I have not tested enough to find hide problems.
- I have not tested all the forms.
- eigen_tests, graspit_collision_tests and transform_test are working. primative_planner_tests crash after the first test.

Some people from here know this work, but it needs a lot of manpower to review and fix a lot of issues. But, it's a first step.  I'm not an expert Graspit user, just want to use it for my work and it was so frustrating to depend on that old versions.

cellolegs, 2021-01-08:
@lepalom Thanks for the info, I'll look into the Qt5 build as I am struggling to compile graspit in ubuntu 20 as the libsoqt4 is no longer included in the qt stacks and so far I haven't found a way to overcome this. For now, I will try compiling in an older distro of ubuntu and see if that helps. 

That said, I was able to overcome my original error which was the same as the poster by adding QtGUI to line 107 in the cmakelists (find_package(Qt4 COMPONENTS QtCore QtSql REQUIRED. Unfortunately that's when I got the errors relating to the SoQT4 lib....

cellolegs, 2021-01-08:
@lepalom thanks for the fork! I compiled beautifully first try! bless you!

When I was struggling with the original one  I was able to overcome my original error which was the same as the poster by adding QtGUI to line 107 in the cmakelists (find_package(Qt4 COMPONENTS QtCore QtSql REQUIRED. Unfortunately that's when I got the errors relating to the SoQT4 lib....which is no  longer in the qt stacks for ubuntu focal :( so looks like the OG graspit is not focal friendly to say the least.

k-ish0212, 2021-12-20:
I got same error above and couldn't install two packages: libcoin80-dev  libsoqt4-dev in the installation tutorial. 
http://graspit-simulator.github.io/build/html/installation_ros.html

enyen, 2022-11-21:
I got the same error, and then I install 
libcoin80-dev, libsoqt4-dev and their dependencies (whenever requested)
manually from https://packages.ubuntu.com/bionic/libcoin80-dev and so on.