graspit-simulator/graspit #162: GUI goes wrong occasionally

zhangcogito, 2019-12-02:
Hello,
I had installed the graspit-ros package and run the graspit-simulator successfully, like this:

[image]

But sometimes the GUI couldn't display correctly, like this

[image]

And it could not also load the world file after I click Menu->open. The window showed noting.

[image]

Then I closed the graspit-simulator and opened again. If luckily, after I repeated many times, the GUI will display correctly.
Is there anyone who has encountered the same problem. How can I resolve it? (I run the simulator in Ubuntu18.04)

iretiayo, 2019-12-02:
@zhangcogito Do you get an error on the terminal? What is printed out on the terminal where you run graspit_simulator?

zhangcogito, 2019-12-02:
> @zhangcogito Do you get an error on the terminal? What is printed out on the terminal where you run graspit_simulator?

Thanks for your reply!
There is no output in the terminal.

[image]

But now, it gets better. If I wait for several minutes and then run the graspit-simulator, the GUI will display correctly.