graspit-simulator/graspit #163: EigenValues in hands models

yruoxi, 2019-12-04:
Hi, Can you give some explanation of the eigenvalue in robot hand models? There isn't much information explained in the user manual.

Humanhand for example, there are multiple eigenvalue parameters, what does 2-dimensional subspace with two eigengrasps mean? and does origin means the open joint state of the hand?

<?xml version="1.0" ?>
<EigenGrasps dimensions="20">
	<EG>
		<EigenValue value="0.51"/>
		<Limits min="-4.0" max="4.0"/>
		<DimVals d0="-1.0" d1="1.0" d2="0.3" d3="0.15" d4="0.0" d5="1.0" d6="0.0" d7="0.0" d8="1.0" d9="1.0" d10="0.5" d11="0.15" d12="1.0" d13="1.0" d14="1.0" d15="0.3" d16="-1.0" d17="1.0" d18="0.1" d19="0.1"/>
	</EG>
	<EG>
		<EigenValue value="0.25"/>
		<Limits min="-4.0" max="4.0"/>
		<DimVals d0="0.0" d1="-1.0" d2="1.0" d3="0.3" d4="0.0" d5="-1.0" d6="1.0" d7="0.3" d8="0.0" d9="-1.0" d10="1.0" d11="0.3" d12="0.0" d13="-1.0" d14="1.0" d15="0.3" d16="-1.0" d17="1.0" d18="0.0" d19="0.0"/>
	</EG>
	<ORIGIN>
		<EigenValue value="0.0"/>
		<DimVals d0="0.131492" d1="0.742900" d2="0.802884" d3="0.461073" d4="0.000000" d5="0.811000" d6="0.726842" d7="0.414373" d8="-0.177118" d9="0.705853" d10="0.611020" d11="0.465138" d12="-0.135235" d13="0.576555" d14="0.191771" d15="0.287653" d16="0.015201" d17="-0.789654" d18="0.753871" d19="0.315313"/>
	  
	</ORIGIN>
</EigenGrasps>

JenniferBuehler, 2019-12-05:
There is a bit of explanation [on this wiki](https://github.com/JenniferBuehler/graspit-pkgs/wiki/urdf2graspit#eigengrasp-notes) , but if you want to go into detail it's best to go through the list of publications.

yruoxi, 2019-12-07:
Thanks, I will go through the list of publications, the wiki explains some, but I will need more details for 2 sets of Eigen parameters.