graspit-simulator/graspit #174: Qt5 version

lepalom, 2020-07-09:
I have seen that graspit has in its dependencies all refered to Qt4:

sudo apt-get install libqt4-dev
sudo apt-get install libqt4-opengl-dev
sudo apt-get install libqt4-sql-psql
sudo apt-get install libsoqt4-dev

Qt4 is not longer active neither provided by any modern Linux distro, so graspit will be unusable soon. There's any plan to migrate to Qt5?

vladi315, 2022-05-10:
Are there any updates or workaround on this matter? Can anyone recommend alternatives to graspit on Ubuntu 20.04?

DavidWatkins, 2022-05-29:
Use branch qt5, then use the following commands to build graspit:
```bash
sudo apt install -y cmake libsoqt520 libsoqt520-dev libblas-dev liblapack-dev libqhull-dev libeigen3-dev git
sudo apt update
mkdir $HOME/dependencies/
cd ~/dependencies
git clone https://github.com/graspit-simulator/graspit.git -b qt5
cd graspit
export GRASPIT=$PWD
mkdir build
cd build
cmake ..
make -j$(nproc)
sudo make install
```

harishbalasub, 2025-02-02:
Is graspit simulator still being maintained?

I built the QT5 version as stated above. 
I ran the graspit_simulator, opened the world 'dlr_flask.xml', then clicked on 'Auto Grasp' from the menu.
The GUI crashed at 
void CollisionInterface::replaceContactSetWithPerimeter(ContactReport &contactSet)
<<
  if ((exitcode = setjmp(qh errexit))) {
    delete [] array;
    if (qhfp) { fclose(qhfp); }
    qhull_mutex.unlock();
    return;
  }
>>

Any idea why

harishbalasub, 2025-04-16:
> In case it helps someone ([@harishbalasub](https://github.com/harishbalasub) ?), after looking into the issue, I found the cause.
> 
> There is a discrepancy between the preprocessor flags of QHull and of graspit. In `graspit/CMakeMacros/FindQhull.cmake`, commenting out the section
> 
>   if(NOT QHULL_USE_STATIC)
>     add_definitions("-Dqh_QHpointer")
>     if(MSVC)
>       add_definitions("-Dqh_QHpointer_dllimport")
>     endif(MSVC)
>   endif(NOT QHULL_USE_STATIC)
> removes the faulty preprocessor flag. The GUI works fine for me after this change.

Oh thats good to hear. When I tried it in Feb, the GUI was crashing in different places and I gave up.