graspit-simulator/graspit #186: graspit_interface can't Import Robot on ROS-neotic, ubuntu 20.04

RyanJiao, 2023-03-01:
Hi, I followed the installation as @DavidWatkins instructed in this issue: https://github.com/JenniferBuehler/graspit-pkgs/issues/62 . And I could suceessfully run the graspit_interface.launch. However, the gui will stop working when I click the File -> Import Robot button.

The gui will color will get a little bit darker than before. Is this something wrong with the qt5?
[image]

zp2546265641, 2024-07-01:
have you solved? why no one here? i think i likely see you in many scenes. strong you are! do you do five fingerhands simulation in gazebo? i have a problem :
[ERROR] [1719833344.944049040]: Group 'l_index' has a mimic joint. Will not initialize dynamics solver
[ERROR] [1719833344.944094085]: Group 'l_middle' has a mimic joint. Will not initialize dynamics solver
[ERROR] [1719833344.944109945]: Group 'l_pinkly' has a mimic joint. Will not initialize dynamics solver
[ERROR] [1719833344.944119533]: Group 'l_ring' has a mimic joint. Will not initialize dynamics solver
[ERROR] [1719833344.944127728]: Group 'l_thumb' has a mimic joint. Will not initialize dynamics solver
[ERROR] [1719833344.944257012]: Group 'r_index' has a mimic joint. Will not initialize dynamics solver
[ERROR] [1719833344.944266710]: Group 'r_middle' has a mimic joint. Will not initialize dynamics solver
[ERROR] [1719833344.944276679]: Group 'r_pinkly' has a mimic joint. Will not initialize dynamics solver
[ERROR] [1719833344.944283121]: Group 'r_ring' has a mimic joint. Will not initialize dynamics solver
[ERROR] [1719833344.944289653]: Group 'r_thumb' has a mimic joint. Will not initialize dynamics solver

wuwuwu~~~do you konw it?或许大佬知道！我没辙了!wuwu~~~~~~~~~~~~~~~~~

itsmevjnk, 2024-11-14:
Very late reply here, but could it be that the Qt5 branch isn't compatible with the DE you're using? I've been using it on Xfce (i.e. Xubuntu) without any significant issues. Ubuntu 20.04 also works for me too.